matafonoff/J1850-VPW-Arduino-Transceiver-Library #1: Fails to compile

Devon7, 2020-09-18:
```
Arduino > File > Examples > J1850 VPW Arduino Transceiver Library > j1850-to-serial > Verify
Arduino: 1.8.13 (Mac OS X), Board: "Arduino Uno"

In file included from …/J1850-VPW-Arduino-Transceiver-Library/src/pins.h:13:0,
                 from …/J1850-VPW-Arduino-Transceiver-Library/src/j1850vpw.h:14,
                 from …/J1850-VPW-Arduino-Transceiver-Library/examples/j1850-to-serial/j1850-to-serial.ino:2:
…/J1850-VPW-Arduino-Transceiver-Library/src/interrupts.h:12:10: fatal error: functional: No such file or directory
 #include <functional>
          ^~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Arduino Uno.
```

matafonoff, 2020-09-19:
That's strange. Will take a look tomorrow. Probably there are some default
libraries are missing in arduino in macosx

сб, 19 сент. 2020 г., 04:24 Devon Sean McCullough <notifications@github.com

Devon7, 2020-09-19:
$ git checkout 75317e9dbe0618ce0037eac16f1d56dd766886eb
compiles despite 11 warnings.  Now to build the circuit and try it on the car.

```
examples/transceiver/transceiver.ino:24:21: warning: suggest parentheses around assignment used as truth value [-Wparentheses]
src/pins.cpp:169:19: warning: invalid conversion from 'int' to 'PIN_MODES' [-fpermissive]
src/j1850vpw.cpp:168:22: warning: suggest parentheses around assignment used as truth value [-Wparentheses]
src/j1850vpw.cpp:220:52: warning: cannot declare member function 'static void J1850VPW::setActiveLevel(uint8_t)' to have static linkage [-fpermissive]
src/j1850vpw.cpp:234:56: warning: cannot declare member function 'static void J1850VPW::init(uint8_t, uint8_t)' to have static linkage [-fpermissive]
src/j1850vpw.cpp:246:89: warning: cannot declare member function 'static uint8_t J1850VPW::send(uint8_t*, uint8_t, int16_t)' to have static linkage [-fpermissive]
src/j1850vpw.cpp:277:34: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
src/j1850vpw.cpp:341:52: warning: invalid conversion from 'uint8_t {aka unsigned char}' to 'J1850_ERRORS' [-fpermissive]
src/j1850vpw.cpp:344:56: warning: cannot declare member function 'static void J1850VPW::onError(onErrorHandler)' to have static linkage [-fpermissive]
src/j1850vpw.cpp:349:86: warning: cannot declare member function 'static int8_t J1850VPW::tryGetReceivedFrame(uint8_t*, bool)' to have static linkage [-fpermissive]
src/j1850vpw.cpp:49:15: warning: 'RX' defined but not used [-Wunused-variable]

```

Alexchertkov, 2020-11-29:
Hi! 

@matafonoff do you have any updates, I can't compile last commit too.

Thanks

matafonoff, 2020-11-30:
Will check and let you know

laszlodaniel, 2021-01-02:
Is there any update? I have the same issue. Tried including StandardCplusplus.h before j1850vpw.h but still won't compile. Partly because Arduino IDE doesn't agree the ways of StandardCplusplus:
```
[...]Arduino\libraries\StandardCplusplus/functional:459:2: error: expected unqualified-id before 'decltype'
```
And the J1850 library also has something going on:
```
transceiver:12:34: error: cannot call member function 'void J1850VPW::onError(onErrorHandler)' without object
```
Windows 10, Arduino IDE 1.8.12

flyboy013, 2021-01-19:
@laszlodaniel - I'm seeing the same error when compiling the code in Arduino IDE 1.8.13 on MacOS.  Did you make any progress?

`sniffer:12:34: error: cannot call member function 'J1850VPW* J1850VPW::onError(onErrorHandler)' without object`

Devon7, 2021-01-19:
I no longer have that '95 Tercel so I can't test J1850 VPW anymore.
A better design would have pin change interrupts queue
pulse widths for decoding by the main program
or queue octets and non-octet signals,
e.g., queue overrun, BRK, EOD, ...

        Peace
            --Devon

P.S. The old commit works but I suggest redesign from scratch
or at least filter out glitches < 8 μs as required by J1850,
which might explain my CSMA/CR echo delay trouble.

matafonoff, 2021-01-22:
Hello! Please provide your code improvements via pull request.

ср, 20 янв. 2021 г. в 01:34, Devon Sean McCullough <notifications@github.com

>:

> I no longer have that '95 Tercel so I can't test J1850 VPW anymore.
> A better design would have pin change interrupts queue
> pulse widths for decoding by the main program
> or queue octets and non-octet [...]

matafonoff, 2021-01-22:
Fixed. Library will be updated within 48h in the Arduino library.

matafonoff, 2021-01-22:
Fixed in v1.0.1 (v1.0-rc2)

closed 2021-01-22 by matafonoff