matafonoff/J1850-VPW-Arduino-Transceiver-Library #3: Round trip?

Devon7, 2020-10-10:
Has this library been able to establish two-way communication with an OBD2 VPW bus?

J1850VPW::send fails with WRITE J1850_ERR_ARBITRATION_LOST
as passive Tx invariably sees active Rx, even when not connected
so it's only seeing its own output with no possible interference.
Perhaps my implementation of your circuit is slow to change?

I'd delay 8 µs before collision checking
given that SAE J1850 says to filter glitches < 8 µs
according to j1850_wp.pdf _Implementing the J1850 Protocol_ by D. John Oliver of Intel.

matafonoff, 2020-10-11:
Interrupts must be switched off during transmission, so no loop back could
occur.

Devon7, 2020-10-11:
The loop back is in hardware and is an essential feature of J1850 VPW CSMA protocol.

Devon7, 2020-10-11:
Test program measures 0 µs delay between Tx leading edge and consequent Rx
when not connected.  Will update this comment when I'm able to test it on a car
but this suggests my patch cannot fix the collision trouble.

matafonoff, 2020-10-12:
What car are you trying to communicate with?

Devon7, 2020-10-12:
1995 Toyota Tercel 4 door sedan made in Japan, VPW pins on OBD connector.
Not all 1995 cars had OBD2 but this has a certification sticker under the hood.

matafonoff, 2021-01-22:
Sorry for answering late. Had no time for this project.

I was implementing subset of J1850VPW that is enough and applicable for Chrsyler vehicles. Not all features are supported.

closed 2021-01-22 by matafonoff