matafonoff/J1850-VPW-Arduino-Transceiver-Library #4: Not Configurated  error - can you help ?

acln99, 2021-03-05:
Hi - using   "sniffer.ino" with  Mega 2560.     In Arduino   Serial Monitor I  receive the message 
   "WRITE  J1850_ERR_RECV_NOT_CONFIGURATED".

I have checked and checked but can't find anything wrong with circuits.   

Do you have any idea's?

Thank you for taking time to read this !!

Tony Goertz
Canada

flyboy013, 2021-03-05:
According to the source code, this error is only generated in: J1850VPW::send.

The first line of code calls "isReadonly" and generated the specified error if it returns true.   

```
uint8_t J1850VPW::send(uint8_t *pData, uint8_t nbytes, int16_t timeoutMs /*= -1*/)
{
    if (isReadonly())
    {
        return handleErrorsInternal(J1850_Write, J1850_ERR_RECV_NOT_CONFIGURATED);
    }
```

isReadonly function returns true if: __txPin.isEmpty.

```
bool J1850VPW::isReadonly() const {
    return __txPin.isEmpty();
}
```

__txPin is set when you call J1850VPW::init with two arguments,  __txPin being the second.
```
J1850VPW* J1850VPW::init(uint8_t rxPin, uint8_t txPin)
{
    init(rxPin);

    __txPin = Pin(txPin, PIN_MODE_OUTPUT);
    __txPin.write(PASSIVE);

    return this;
}
```

So, did you call init with two arguments?

acln99, 2021-03-05:
Hi – Thank You for the response.  

 

I simply took the   “sniffer.ino”  and tried to execute it.  It compiled OK.

 

This is the sample  sniffer.ino code:

#include "j1850vpw.h"

 

#define TX 9

#define RX 10

 

void handleError(J1850_Operations op, J1850_ERRORS err);

J1850VPW vpw;

 

void setup()

{

    Serial.begin(115200);           // start serial port

    vpw.onError(handleError); // listen for errors

    vpw.init(RX);             // init transceiver

}

 

 

So – would this be the correct  line”

    vpw.init(RX, TX );             // init transceiver

 

 

Thank you  -  I am old  and retired and am now learning something new.   I’m very new to Arduino programing !

 

-- Tony

acln99, 2021-03-05:
Hi – I changed the sniffer.ino code and now get   

 

WRITE  J1850_ERR_ ERR_ARBITRATION_LOST   

I also tried the  sample program in the README.md  file.  

I am connected to a 2002 Chrysler SBEC3A  PCM  with J1850 communication  on the bench and no other modules connected.

 

Again – Thank you for sharing your programs and taking the time to help.

-- Tony

flyboy013, 2021-03-05:
Ok.  I looked at the sniffer example, and it seems to have errors.

```
void setup()
{
 
    < code removed >

    vpw.init(RX);             // init transceiver
}

void loop()
{

    < code removed >

    // write simple message to bus
    vpw.send(sendBuff, sizeof(sendBuff));

    delay(1000);
}
```

This code did an initialization for read-only, but then tried doing a send.

If you are only trying to sniff the network, you should remove the "vpw.send".    Also, be careful about what you are writing to the bus.  I see the example is trying to write "{0x01, 0x02}".  Not sure what impact this may have as I'm not sure if this is a valid command for your vehicle.

Also, the arbitration loss isn't an error per se.  It just means that multiple devices were sending data onto the bus, and your program loss the arbitration based on the J1850 arbitration rules.

laszlodaniel, 2021-03-06:
Thanks @matafonoff for reworking the source code and @flyboy013 for the help so far!

Due to the lack of adequate PCM I'm testing the sniffer example on two Arduino boards (Uno and Mega).
Uno runs part of the example that sends the 01 02 message for every 1 second, and Mega runs part of the example that listens to activity.
I skipped the transceiver circuit and connected the 2 boards directly with wires (RX to TX and TX to RX).
Problem is now that the receiver board does not seem to get the 01 02 message, the Serial Monitor is empty.
Is there something more to be done here? I'm guessing pull-up resistors are needed somewhere.

@acln99 is testing a transceiver development board that I have built for him. I followed the schematic shown on the main page with the following significant differences:
- NPN transistors are replaced with N-channel MOSFETs,
- R14 is 10k, this may limit the bus-driving current too much and render the transmitter part useless.

Please advise.

[image]

matafonoff, 2021-03-06:
Hello!

The logic is quite simple - if you are not using rx and tx pins you can't
send data to j1850 due to some arbitration checks (reading data from j1850)
that must be done during data send process.

Sniffer is ment to be a readonly code that sends nothing to j1850.

And yes, welcome to arduino world :)

acln99, 2021-03-09:
I am stuck  - still no progress.
I have created  the "Receive" portion of the circuits again as per the wiring diagram  .

I am using an Arduino Mega
The sniffer.ino    compiles and does not give any errors  in the monitor.

I also tried  "transceiver.ino"  sample code  with   //     vpw.send(sendBuff, sizeof(sendBuff));    commented out.

I never receive any  "output" in the monitor.

The PCM in on my bench - not in car.
With PCM disconnected,    PCI lead from PCM  to Ground  is  1000 ohms.      ( I believe correct )

With PCM powered up,    Voltage from PCI lead from PCM   to Ground  fluctuated  .54v  to .65v  with spikes to 1.0V 
I'm using a basic digital meter.     While this is not a good test,  this is also expected as per the documentation I have been able to find.

I can connect DRBIII   and monitor for bus activity  which appears to be normal.  
I have tried   2 different  PCM's.

Any thoughts   and/or   help     on  how to look for my mistake is greatly appreciated !!
Thank You

matafonoff, 2021-03-10:
Please contact me via telegram: @XelberaSW

matafonoff, 2021-03-11:
> Thank you for trying to help. I have not been able to send to XelberaSW on
> instagram !
>
>
>
> My email is ***@***.*** or tonygoertz on instagram
>
>
>
> I have kept testing and have confirmed communication on PCI pin #2 on DLC
> using ELM327 BT mini and android apps.
>
>
>
> I can’t find what I am doing wrong !!
>
>
>
> Thank You !

Not instagram, but telegram

matafonoff, 2021-06-14:
It seems that we've got everything work

closed 2021-06-14 by matafonoff