matafonoff/J1850-VPW-Arduino-Transceiver-Library #5: Substitute diode for ZPD8V2 ?

wpuziewicz, 2021-03-27:
Hi,

I'm looking to build this project, but I can't find a source for the ZPD8V2 diode listed in the schematic.  I found a datasheet for it but I'm stumped at the Digikey spec search.  I don't know enough to pick a substitute.  Can you recommend another diode that will work in its place?

Thanks

matafonoff, 2021-03-27:
You need any zener diod with stabilization voltage at around 7.5-8.2v.

wpuziewicz, 2021-03-27:
Thanks for the quick response!

For anyone else with the same question, here is what I ordered from Digikey.  I got a 7.5v and an 8.2v just in case one works better than the other.

1N5344BRLGOSCT-ND‎
‎1N5344BRLG‎
DIODE ZENER 8.2V 5W AXIAL 

1N5343BRLGOSCT-ND‎
‎1N5343BRLG‎
DIODE ZENER 7.5V 5W AXIAL

closed 2021-03-27 by wpuziewicz