matafonoff/J1850-VPW-Arduino-Transceiver-Library #6: READ J1850_ERR_PULSE_OUTSIDE_FRAME | WRITE J1850_ERR_ARBITRATION_LOST

laszlodaniel, 2021-04-17:
Hi, back again with another transceiver circuit:

[image]

The circuit is in standalone mode: PCI-bus (J1962-2) pin is left floating.
SLEEP pin is tied to 5V to enable transmitter (receiver is always enabled).
4XLOOP pin is tied to GND to enable waveshaping. When left floating to enable loop-back mode, TX is fed back to RX after waveshaping without transmitting on PCI-bus, it makes no difference.

Running "transceiver.ino" on an Arduino Uno. No matter which pins I use (both capable of pin change interrupt) I get these error messages repeatedly on the Serial Monitor:

```
READ J1850_ERR_PULSE_OUTSIDE_FRAME
WRITE J1850_ERR_ARBITRATION_LOST
READ J1850_ERR_PULSE_OUTSIDE_FRAME
WRITE J1850_ERR_ARBITRATION_LOST
READ J1850_ERR_PULSE_OUTSIDE_FRAME
WRITE J1850_ERR_ARBITRATION_LOST
...

```

When I disable waveshaping (4XLOOP pin tied to 5V) then I get this error message only once:

```
READ J1850_ERR_PULSE_OUTSIDE_FRAME
```

I don't have an oscilloscope or logic analyzer for troubleshooting.
RX/TX pins seem to stay at 5V and PCI-bus pin at 7V without changing.

When I manually toggle the TX-pin:
```
TX: floating, RX: 0V,   PCI: 370mV
TX: 0V,       RX: 0V,   PCI: 370mV
TX: 5V,       RX: 4.7V, PCI: 7V
```

Please advise, what am I doing wrong?

matafonoff, 2021-06-14:
Have never tried working with external transceivers.

Second thing is getting the data with oscilloscope trying to understand if everything is OK in message. Otherwise this is a finding a needle in a haystack.

laszlodaniel, 2021-06-19:
Couple of things I have learned since then. This might narrow down the haystack.

On Arduino Mega the library doesn't enable pin change interrupts automatically. I have to do it manually after initializing the library:

```
sbi(PCICR, PCIE0); // PCINT7:0 pins
sbi(PCMSK0, PCINT4); // enable PCINT4 pin change interrupt (D10)
```

On Arduino Uno there's no such issue, the library inits everything right.
Your library seems to handle this like it should, still somehow it doesn't like the Mega.

Since I don't have an oscilloscope or logic analyzer, I did the best I could and made a small sketch that records when the RX pin changes state:

```
Message: 01 02 C8

Timestamp    -> micros   | difference
-------------------------------------
18:01:27.433 -> 24211996
18:01:27.433 -> 24212084 | 88
18:01:27.433 -> 24212184 | 100
18:01:27.433 -> 24212340 | 156
18:01:27.433 -> 24212436 | 96
18:01:27.433 -> 24212600 | 164
18:01:27.433 -> 24212696 | 96
18:01:27.433 -> 24212860 | 164
18:01:27.433 -> 24212960 | 100
18:01:27.433 -> 24213060 | 100
18:01:27.433 -> 24213160 | 100
18:01:27.433 -> 24213320 | 160
18:01:27.433 -> 24213416 | 96
18:01:27.433 -> 24213576 | 160
18:01:27.433 -> 24213672 | 96
18:01:27.433 -> 24213832 | 160
18:01:27.433 -> 24213996 | 164
18:01:27.433 -> 24214164 | 168
18:01:27.433 -> 24214328 | 164
18:01:27.433 -> 24214424 | 96
18:01:27.433 -> 24214520 | 96
18:01:27.433 -> 24214680 | 160
18:01:27.433 -> 24214840 | 160
18:01:27.433 -> 24215004 | 164
18:01:27.433 -> 24215100 | 96
18:01:27.433 -> 24215260 | 160
```

The first time difference is always less than RX_SOF_MIN (163 us) that's why I get the `READ J1850_ERR_PULSE_OUTSIDE_FRAME` error.
Even though the library explicitly waits 200 us in a while loop, for some reason it does not actually. How could this be?

The long/short pulses are also on the edge, perhaps due to the inherently inaccurate micros() measurements in the loop().

matafonoff, 2021-06-19:
Thank you for your investigation.

Have no mega to do tests at the moment, so could not do a fix for interrupt
enabling. Feel free to provide a PR.

About the timings - will check what's going on there within the week.

matafonoff, 2021-06-19:
Feel free to contact me via telegram (@XelberaSW) - this could make things
done faster.

closed 2023-08-09 by matafonoff