matafonoff/J1850-VPW-Arduino-Transceiver-Library #15: some basic questions regarding the library and the interface

santon74, 2022-05-27:
Hello,
I am trying to implement this library and the interface for communicating with early 2000 GM truck.  I have some rather basic questions regarding the interface circuit and the library.  I hope it is OK to ask them here.
1) The author stated that the library was designed for Chrysler vehicle.  Will this library work with J1850 VPN of GM?  
2) I noticed that the 3 transistor interface circuit will tie the J1850 bus line to the HIGH state.  Is it correct?  If so, could it cause any issue?
3) is there a simple way to simulate the J1850 VPN signals without using the car's components on the bench?  The second Arduino, for example?
Sorry for these questions- I am a complete beginner in this field.
Thanks!

matafonoff, 2022-05-28:
Hello!

1. It must be working with any j1850vpw, but was tested hardly on Chrysler/dodge vehicles. 
2. Yes, so the circuit must be powered before vehicle startup to prevent bus communication issues
3. You could use 2 arduonos . I’ve tested this scenario before working with vehicle’s data bus

Best regards, 
Stepan Matafonov!

santon74, 2022-05-28:
Dear Stepan,
Thank you for your prompt response.
I tried to use 2 Arduinos, connected directly, without the interface circuit (Rx to Tx, Tx to Rx).  On one of them, I used a "transceiver.ino" or "j1850-to-serial.ino" sketch while another one was running "sniffer.ino" sketch. With this arrangement, I was getting the ERR_PULSE_OUTSIDE_FRAME error.  Is there anything that I am doing wrong?
Thanks.
Anton.

matafonoff, 2022-05-28:
It does not work without circuit. Don’t know why, but I had this issue

Best regards, 
Stepan Matafonov!

closed 2023-08-09 by matafonoff