matafonoff/J1850-VPW-Arduino-Transceiver-Library #16: Error Using MEGA with Harley

1234567890aaaaa, 2022-06-09:
I'm struggling to get information from my Harley ECU using any of the three sketches. I read that Harley's use 10.4 kbps for their J1850 VPW communication, so in the J1850-to-Serial sketch I changed the default to 10400 (after the Serial.available > 0).
Commonly I get the "WRITE J1850_ERR_ARBITRATION_LOST".
I tried to put in "Serial.println("Setting up new ports");" in the Serial.available case switching, but the Arduino never sees Serial.available > 0.

Honestly I'm not sure what the difference is between the three sketches, and which one I should be using to just read from my data port.

Thanks in advance.

matafonoff, 2022-06-09:
Setting serial speed makes no sense - it does not affect j1850 communication speed.
From the other hand, using lower baud rate makes uC waste more time on serial communication interrupts. 

My suggestions:
- set serial baudrate above 57kbs
- check schematics (especially transistors pin out). I had this kind of an issue when one of my transistor was dead.

Best regards, 
Stepan Matafonov!

TaylorRL95, 2022-06-09:
I see. I thought each manufacturer had their own Vaud rate for the cow signals, and that’s what the dual Serial.available was handing.
I got it to work by switching to an Uno. I read somewhere that the Mega pins aren’t configured for interrupts by default and the Uno are. It was a little confusing on the circuit diagram whether each Out / In are TX or RX.
Thanks for the help and the library!

closed 2022-06-09 by TaylorRL95

matafonoff, 2022-06-09:
I’m glad it’s working for you! Have a nice day!

Best regards, 
Stepan Matafonov!