matafonoff/J1850-VPW-Arduino-Transceiver-Library #19: First use on ESP32 chrysler radio

wait-youre-married, 2023-08-09:
Hello, couldn't find nothing too specific, you mentioned being able to connect to a chrysler radio running off of the J1850 VPW bus. Could I ask what was the original project, as I'm trying to add A2DP and AVRCP bluetooth capability to my chrrysler radio which also runs off of the J1850 bus. I noticed the schematic for the circuit to connect to the actual bus and read snd write messages, i bought an Arduino ESP32 and what I want to do is have the ESP32 send constant messages on the bus to let the radio know it is there (masked as a cd changer), but also want it to monitor the bus to see the messages generated when TRACK UP or DOWN buttons are pressed, RANDOM button pressed and so forth to also integrate hands free calling. I did see the pull request for the PCB with the components on it, but there is no link to buy it. I know there are products on the market that will do exactly what i asked, problem is I tried one of them and it didnt work, and they are quite costly, I think the problem was that it was sending an idle message way too frequently and I also think it may not have been exactly the right message.

matafonoff, 2023-08-09:
Hello! Unfortunately, ESP32 is not supported at the moment, just atmega328-based Arduinos. If you need to add support of you board, you should extend interrupts and pins classes. These classes are hardware-specific pin access layer that is several times faster than standard Arduino's ones. Execution time is extremely critical for VPW modulation.

wait-youre-married, 2023-08-09:
Hello, thank you for your reply. This might be a very silly question (I've started my journey with arduino quite literally a couple of days ago by buy the ESP32, I still have no idea how to code anything). But could the atmega328 board be made to interface with the ESP32 such as that the atmega would read the BUS and the ESP32 would just read the actual numerical values produced by the atmega board. The same values you would see when directly communicating on the bus using an elm327 obd interface. And then these numerical values be paired with the various AVRCP commands.

matafonoff, 2023-08-09:
sure! you could use serial port. And yes< sometimes it's much easier to communicate with elm327 with AT commands directly to get rid of all j1850-related stuff

Lorenzch, 2023-08-22:
Hello. I'm planning to install the original parking sensors out from a Chrysler Grand Voyager, but that involves several modules apart from the parking sensor module itself. In order to work I need an ECM, TCM, BCM and an Instrument Cluster. I made it to work in a bench but I don't find logical to install and hide inside the car soo many modules, so I was wondering if I can use the arduino to read all of the PCI-Bus signals and just have the arduino reproducing them instead of having to connect all of the modules to the Bus network.  
I'd like to congratulate you for this project. Thanks.

matafonoff, 2023-08-22:
Hello! Yep, it could be done. I believe main messages are engine state and shifter position messages. The module must work using these messages according to the logic

Lorenzch, 2023-08-22:
I'm really glad to hear that, because I was about to give up on this project. I'm not a dumb with computers, but I've never used an Arduino. ¿would you be willing to help me in reading the messages and emulate them? Thanks.

matafonoff, 2023-08-28:
> <matafonoff/J1850-VPW-Arduino-Transceiver-Library/issues/19/1688974785@
> github.com>

There is an example of the code in the readme file. Please feel free to use
it

Lorenzch, 2023-09-02:
Yes, I've read. I have two questions. The readme file includes an image with a circuit with several transistors and resistances; for my project would I need have that circuit or the installation would be plug and play between the car's pci bus and the arduino?

The other question is that I noticed the module also needs to constantly knows vehicle speed as to disable the front sensors when you're not stopped or at parking speeds; that means that the arduino wouldn't just be sending a constant message to the bus that I could record and the arduino reproduce, but, It would have constant different messages based upon the speed.
Thank you.

matafonoff, 2024-01-03:
Hello! You need the level shifter schematics provided in README file. PCI bus works at 8 volts approx - this will kill your arduino without level shifter. You could use any logical level shifter that could switch it's state in 2 microseconds at least and could convert 8v signals to 5v ones and vice versa.
About the speed: try searching info for [0x10: Engine RPM, Speed, MAP] message in j1850 docs.

closed 2024-01-03 by matafonoff